cansik/realsense-processing #2: problems with gradle build on OSX

dongbeikezhenxi, 2018-12-22:
Hello, trying to get this to build and work on macOS 10.14.1, Java 1.8.0_191, following the steps, and I get this error (the librealsense was successfully built from cansik/librealsense):

```
> Task :nativeJavah FAILEDi

FAILURE: Build failed with an exception.

* What went wrong:
A problem was found with the configuration of task ':nativeJavah'.
> Value '[/Users/yuhaomao/librealsense/wrappers/java/build/classes/main/jar]' specified for property '$1' cannot be converted to a directory.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1s
2 actionable tasks: 1 executed, 1 up-to-date"
```

dongbeikezhenxi, 2018-12-27:
For this build to succeed had to use `./gradlew` on OS X, different gradle version from brew's.

cansik, 2018-12-27:
Ok I can reproduce this on MacOSX Mojave. I think it is a dependency problem. It is working with gradlew?

cansik, 2018-12-27:
Here is the gradle build error:
https://scans.gradle.com/s/5h372m5j4vsf4/