cansik/realsense-processing #3: for processing2

dongbeikezhenxi, 2018-12-23:
Hello, trying to get this to build and work on macOS 10.14.1, Java 1.8.0_191, following the steps, and I get this error on processing2 .(btw it works on processing 3)
```
No library found for org.librealsense.Native
No library found for org.librealsense.Native
No library found for org.librealsense.Native
This version of Processing only supports libraries and JAR files compiled for Java 1.6 or earlier.
A library used by this sketch was compiled for Java 1.7 or later, 
and needs to be recompiled to be compatible with Java 1.6.

```

cansik, 2018-12-23:
The focus was not on implemented a generic library. It was only tested with processing 3.

dongbeikezhenxi, 2018-12-26:
Thank you. It seems I only need to retarget the built 1.6 for to make it work, but somehow all my binaries  and .class files are coming out as Java SE 1.8 (major version 52). I tried setting -target 1.6 and targetCompatibility to 1.6 to no avail in `build.gradle` in librealsense/wrappers/java...

Below is my current build.gradle:

```
buildscript {
    repositories {
        mavenCentral()
    }
}

plugins {
    id 'jvm-component'
//    id 'java'
    id 'java-lang'
//    id 'java-library'
    id 'cpp'
    id 'maven-publish'
    id 'org.bven.jniplugin' version '0.0.5'
    id 'idea'
}

version = '2.15.0-0'
group = 'org.librealsense'

// Process resources task doesn't get created if there aren't any resources; create a placeholder to ensure that it does.
def placeholder = file('src/main/resources/deleteme')
if (!placeholder.parentFile.exists()) {
    placeholder.parentFile.mkdirs()
}
if (!placeholder.exists()) {
    placeholder.createNewFile()
}
placeholder.deleteOnExit()

apply plugin: 'maven-publish'

//allprojects {
//apply plugin: 'java'
//apply plugin: 'idea'
//apply plugin: 'java-lang'
//apply plugin: 'java-library'
//sourceCompatibility = 1.8
//targetCompatibility = 1.6
//}

tasks.withType(JavaCompile)
{
    //apply plugin: 'java'
    //options.compilerArgs += ["-target 1.6"]
    sourceCompatibility = "1.6"
    targetCompatibility = "1.6" 
    println 'Compiler args: ' + options.compilerArgs
}

model {
    platforms {
        ['windows', 'linux', 'osx', 'freebsd-10', 'solaris'].each { os ->
            ['x64'].each { arch ->
                "$os-$arch" {
                    architecture arch
                    operatingSystem os
                }
            }
        }
    }

    components {
        main(JvmLibrarySpec) {
        }

        'native'(JNILibrarySpec) { jniLibSpec ->
            library "main"
            nativeClass "org.librealsense.Native"

            $.platforms.all {
                jniLibSpec.targetPlatform it.name
            }

            binaries.all {
                lib new JNIDependency(project)
                cppCompiler.args "-I${project.projectDir}/../../include"
                linker.args "-L${project.projectDir}/../../"

                if (targetPlatform.operatingSystem.macOsX) {
                    //cppCompiler.args "-I${project.projectDir}/../../include"
                    cppCompiler.args "-I${project.projectDir}/../../include"
                    linker.args "${project.projectDir}/../../build/Debug/librealsense2.dylib"
                }

                if (targetPlatform.operatingSystem.windows) {

                    // https://github.com/couchbase/couchbase-lite-java-native/issues/23
                    //cppCompiler.args "/IC:/Program Files (x86)/Windows Kits/10/Include/10.0.10586.0/ucrt"
                    def arch = {
                        if (targetPlatform.architecture.name.contains('arm')) {
                            'arm'
                        } else if (targetPlatform.architecture.name.contains('64')) {
                            'x64'
                        } else {
                            'x86'
                        }
                    }()

//                    linker.args "/LIBPATH:C:/Program Files (x86)/Windows Kits/10/Lib/10.0.10586.0/ucrt/${arch}"
                    cppCompiler.args "-I${project.projectDir}/../../include"
                    linker.args "-lrealsense2"
                    linker.args "/LIBPATH:${project.projectDir}/../../Debug/"
                    linker.args "realsense2.lib"
                }
            }
        }
    }

    tasks {
        run(JavaExec) {
//compileJava {
//
//}
            def compile = $.tasks.compileMainJarMainJava;
            def jar = $.tasks.createMainJar;
            dependsOn $.tasks.mainJar
            classpath compile.classpath
            classpath jar.outputs
            main = 'org.librealsense.Native'
        }

        wrapper {
            gradleVersion = '4.7'
        }
    }

    publishing {
        publications {
            impl(MavenPublication) {
                groupId 'org.librealsense'
                artifactId 'realsense-jvm'
                artifact $.tasks.createMainJar
            }
        }
    }
}
```

The command `./gradlew build` succeeds and so does publish but the .class files are still for Java 8... This is on OS X, gradle 4.7, gradle 5 does not seem to work per compile issue #2

dongbeikezhenxi, 2018-12-27:
While succeeded rebuilding the Java wrapper for the 1.6 target platform, getting a NullPointerException (in either Processing 3 or Processing 2) with the following line:

```
    this.context = Context.create();
```

Can't seem to retrace where it may be failing. Trying to use .dylibs from either pre-built jars with 1.6 Java or freshly compiled .dylibs give issues. Using your prebuilt .jar, works. Can't seem to figure out why, and where the native code generates the exception...

To successfully build the 1.6 target used this in gradle.build:

```
    options.compilerArgs += ["-source", "1.6", "-target", "1.6"]
```

smokhov, 2018-12-27:
I'd like to second this issue. There is some kind of disconnect when recompiling from sources librealsense/wrappers/java with 1.6 target and dylibs, being unable to create Context in processing due to some bizarre NullPointerException, unclear where it is coming from.

cansik, 2018-12-27:
I believe it has to do with the problem that the build is broken on MacOS Mojave (#2) I can not build it anymore too, but I don't have the time right now to investigate.

smokhov, 2018-12-27:
Thank you Florian for looking into this.

Originally, in #2 we rebuilt using `./gradlew`, which pulled in 4.7, and it was successful. After that, we copied `main.jar` replacing `realsense.jar` under the realsense-processing's `lib/` and `examples/RealSenseViewer/code`. The .pde code fails to run when trying to create the `Context` instance with a NullPointerException. We've now rebuilt using gradle 4.9 (it succeeds), and the exception result is the same (in either Processing 3 or 2). Not sure why context creation fails in the recompiled code.

Your prepackaged binaries (jar and dylibs) from August work however in Processing 3 to get both color and depth. Our recompiled wrapper and binaries don't work anywhere having the exception I mentioned. Our recompiled Java wrapper doesn't work with your dylibs either.

We also succeeded rebuilding with gradle realsense-processing project itself, but unsure of its purpose with all the Katlin stuff and the RealSenseProcessing .jar it creates and how it is meant to interact with realsense.jar and .pde files...

smokhov, 2018-12-27:
As a data point, we tried reproducing pure original build using Java 1.8/Processing 3 only for the Java wrapper and the Processing library, and get the same NullPointerException in `Context.create()` inside the Processing example. Kind of stuck here. Build succeeds, runtime doesn't... :(

cansik, 2018-12-28:
Have you checked if the release libraries are in the jar? They are not automatically bundled! To be bundled you have to add them to this folder:

```
librealsense/wrappers/java/src/main/resources/lib/org.librealsense/osx-x64/
```

Then they should be (after compilation) in `realsense.jar/lib/org.librealsense/osx-x64/`:

[image]

This repository is just a proof of concept, using the prebuilt library within java & processing. The code here is written in kotlin, but it does not bias the wrapper.

smokhov, 2018-12-29:
No stacktrace was produced, but I've found the problem. :) Addressing it locally and recompiling now made it work in Processing 2 and 3. Will post more details a bit later.

smokhov, 2018-12-29:
... and so the PR https://github.com/cansik/librealsense/pull/1 fixes the issue that was introduced [here](https://github.com/cansik/librealsense/commit/03ac5dd48a01d0c7d92bb2a689fc9ed85a13e187#diff-44cb992d3606489c42e4a7159189c702R25). The line at fault was:

```java
boolean isRSDEV = System.getProperty("rsdev").equals("1");
```

It assumes `rsdev` is always defined. Unfortunately was not by default true; the property was `null` for us, so `equals()` barfs out nullpointer... Found this through eyeballing the code tracing it back. Hard to catch either since it's in the `static` initializer. Anyway, we are now happy campers and the recompiled lib works in Processing 2 and 3. Thank you for your work!

Perhaps a PR upstream to @edwinRNDR is also in order afterward?

Happy holidays!

cansik, 2018-12-29:
Great, it's merged! Yes, the static constructor is not the best idea, I think we need a new method to load the dynamic libraries, which the user itself can call.

closed 2019-01-04 by cansik