cansik/realsense-processing #4: Issue running RealsenseViewer

lvitters, 2019-01-07:
```
A library used by this sketch is not installed properly.
A library relies on native code that's not available.
Or only works properly when the sketch is run as a 32-bit application.
```

Error message when running RealsenseViewer.pde, line 31 gets highlighted:

`context = Context.create();`

jav-solo, 2019-01-31:
I have this same question, I am getting the same error. I have installed Processing 3 and my OS is Ubuntu  16.04

jav-solo, 2019-01-31:
It actually seems that you can only use the java wrapper for librealsense on macOS and Windows accordng to [this](https://github.com/cansik/librealsense/tree/master/wrappers/java).

cansik, 2019-02-02:
It could work on linux too, but you would have to build the realsense source by yourself and add the right built binaries to the native folder. 

Please be aware that this is **not** a library for processing. It was just a proof of concept which could be the base for a future library.

smokhov, 2019-02-02:
> Please be aware that this is not a library for processing. It was just a proof of concept which could be the base for a future library.

(Un?)fortunately, it seems to be slowly becoming one! ;)

cansik, 2019-02-02:
@smokhov I am working on it ;) => https://github.com/cansik/realsense-processing/tree/poc-to-library

cansik, 2019-02-02:
@lvitters & @jav-solo Please check the new built libraries. I've completely rewritten the java interface and updated to version 2.17.0 of the Intel RealSense api. There is also a [prebuilt binary](https://github.com/cansik/realsense-processing/releases/tag/v2.17.0.1) for unix. Would be great if you could give me feedback if it's working now for you.

jav-solo, 2019-02-04:
Should I put these files somewhere first? I tried to run the examples from the unzipped directory and I get this: `No library found for ch.bildspur.realsense
Libraries must be installed in a folder named 'libraries' inside the sketchbook folder (see the Preferences window).
The package “ch” does not exist. You might be missing a library.`

jav-solo, 2019-02-04:
It looks like the main reason for not being able to run this is this error: `UnsatisfiedLinkError: org.librealsense.Native.rs2CreateContext(I)J
A library relies on native code that's not available.
Or only works properly when the sketch is run as a 32-bit application.
UnsatisfiedLinkError: org.librealsense.Native.rs2CreateContext(I)J`

cansik, 2019-02-05:
On a freshly installed ubuntu, it works to run the simplest example:
[image]

Please make sure to not use the **latest** build, but the one which is [bundled with the unix libraries](https://github.com/cansik/realsense-processing/releases/tag/v2.17.0.1)! Because I did not want to release the unix bindings (100 mb), without testing it.

cansik, 2019-02-05:
Reading the color stream & depth stream is also working :)

[image]

[image]

cansik, 2019-02-07:
@lvitters It was working on my test system on Windows 10. Is the windows-x64 library bundled in the release you are using?

lvitters, 2019-02-07:
> @lvitters It was working on my test system on Windows 10. Is the windows-x64 library bundled in the release you are using?

No, the **native** folder only says osx-x64. Where do I get the windows-x64 library? 

Also, since it didn't work on Mac either, would you mind walking me through how to correctly place which files in Processing's libraries folder?

cansik, 2019-02-07:
@lvitters There was a lot of activity on the repository, so you have to redownload the [latest contributed release](https://github.com/cansik/realsense-processing/releases/download/contributed/RealSenseProcessing.zip).

Installing the library can be done as described here: https://github.com/processing/processing/wiki/How-to-Install-a-Contributed-Library#manual-install

lvitters, 2019-02-07:
Thank you, got it working now!

[image]

closed 2019-02-07 by cansik

jav-solo, 2019-02-07:
So just to be clear, we are to put the RealSenseProcessing directory in the libraries folder of Processing, correct?

lvitters, 2019-02-07:
> So just to be clear, we are to put the RealSenseProcessing directory in the libraries folder of Processing, correct?

Yes, exactly. Make sure to use the one cansik linked in his last comment.

jav-solo, 2019-02-07:
Got it, it is working fine now. Thank you!