cansik/realsense-processing #5: Is there a way to get raw data from RealSense?

jav-solo, 2019-02-01:
I was wondering if there is a way to get the raw data from the real sense using this wrapper?

cansik, 2019-02-02:
What do you mean by the "raw" data? 

Check out the [RealSenseCamera](https://github.com/cansik/realsense-processing/blob/master/examples/RealsenseViewer/RealSenseCamera.pde#L128) class which handles the realsense context.

cansik, 2019-02-02:
Check out the latest released library version. It now supports to read the [raw depth buffer](https://github.com/cansik/realsense-processing/blob/master/examples/UseDepthBuffer/UseDepthBuffer.pde#L26). Is that what you were looking for?

jav-solo, 2019-02-04:
Thank you for replying. As an example, the [Kinect](https://shiffman.net/p5/kinect/) has raw data points that can be used and visualized in Processing using OpenKinect and freenect. I was wondering whether or not there was a difference between the points seen in this poc and what might be considered "raw" points that the realsense camera sees.

jav-solo, 2019-02-05:
Yes, that is what I mean. I read [here](https://forums.intel.com/s/question/0D50P0000490RrZSAU/raw-preprocessed-data-intel-d415?language=en_US) that end users could not access it. I do not fully understand how the camera produces the point cloud.

cansik, 2019-02-05:
@jav-solo Your referenced thread is about getting the [raw pixel data](https://en.wikipedia.org/wiki/Raw_image_format) (direct sensor data) to develop the picture yourself.
And no, that is not possible currently.

jav-solo, 2019-02-06:
I understand, thank you.

closed 2019-02-06 by jav-solo