cansik/realsense-processing #10: Add post processing support

cansik, 2019-02-11:
[image]

ECHO960, 2019-04-29:
Hi, thanks for the code!

I am wondering if there is a way to access the advanced control parameters during the post processing?
Or is there an example code of post processing?

To be specific, we are trying to tune the "DS neighbor threshold" parameter to make the Depth image more accurate. It would be great if you can help! thanks!

cansik, 2019-04-29:
No I think at the moment it is not possible to set this parameter through the java and it won't be supported until Intel adds it to the C API. A [quick search](https://github.com/IntelRealSense/librealsense/search?q=neighbor+threshold&unscoped_q=neighbor+threshold) showed me that this is only possible through the C++ API, right?

cansik, 2019-04-29:
As a guidance, you would have to add the new methods to the following file (and include the new header):
https://github.com/cansik/librealsense/blob/master/wrappers/java/src/rsjvm/cpp/rsjvm.cpp

Then in java you could add the functionality here:
https://github.com/cansik/librealsense/blob/master/wrappers/java/src/main/java/org.librealsense/Native.java#L99

And finally create a class which then executes this native method. Of course, the enable advanced mode method must be implemented too.

ECHO960, 2019-04-29:
thanks for your quick response!
It will be great if you can add it in the future! And I will try to figure it out on my own in the meantime. 
good luck with your thesis!

cansik, 2020-01-07:
@ECHO960 If you are still interested in using the RealSense library from java, check out my latest wrapper project which ports the C# API to Java. Not all 2.29.0 features are supported yet, but most of the common ones are. There is also a test case on how to use it together with processing and advanced controls / settings are supported.

https://github.com/cansik/librealsense-java

oori, 2020-02-01:
@cansik would be great to see the docs updated accordingly.       I couldn't manage to get this working with new java lib (w/ processing).       I managed to run the new snapshop build (and update the examples), but not post-processing, like Colorizer example.         Thanks.

cansik, 2020-02-01:
@oori Yes, this can be done with the new lib. Check out [how I set the option](https://github.com/cansik/librealsense-java/blob/master/src/test/java/processing/ProcessingViewer.java#L50-L51) for the colorizer (would be the same manner for the Threshold filter).

Which docs are you talking about? At the moment, the new wrapper-lib is not used by the processing library because it's still a bit of work to do to create the API.

oori, 2020-02-01:
I built the java lib and use that snapshot in the processing library.    that's fine..  just the "set options" isn't totally clear yet.     I saw you added issue #14,  I will track your updates there.
Thanks again!

cansik, 2020-02-01:
I spent some hours today to update the processing wrapper to enable support for filters / processors and multiple streams. It's already working but not released. The min/max threshold is also implemented (even with settings). Check out following example:

https://github.com/cansik/realsense-processing/blob/master/src/test/java/ch/bildspur/realsense/test/SimpleFilterTest.java#L51

If you run it with your snapshot, use following methods to set the min and the max values:

```java
ThresholdFilter filter = new ThresholdFilter();
filter.getOptions().get(Option.MinDistance).setValue(0.0); // 0 meters
filter.getOptions().get(Option.MaxDistance).setValue(2.0); // 2 meters
```

cansik, 2020-02-14:
Released version 2.1.0 to the contribution manager today.

closed 2020-02-14 by cansik