cansik/realsense-processing #11: missing common getters in RealSenseCamera

smokhov, 2019-03-15:
@cansik -- in the new library, compared to the pre-library version, can't get width, height, FPS, etc. as well as selectively enable streams having things like `getHeight()` or `enableDepth()`...

cansik, 2019-03-15:
What you mean is the **proof of concept**, which directly used the jvm-wrapper for the realsense. But that wasn't a library, but a proof of concept project.

Now it's a valid processing library, hiding a lot of unnecessary things for people who just want to read the different streams. So to continue work with the realsense API, you have to use just the jvm-wrapper which is also included in the release of this library.

smokhov, 2019-03-15:
I guess this is matter of convenience:

- [ ] [E.g., these lines](https://github.com/cansik/realsense-processing/blob/067a758efc00abe62936bcc626dc26fe74ed79ce/src/main/java/ch/bildspur/realsense/RealSenseCamera.java#L37) could use for convenience when accessing the object and its parameters

We are now using the proper library and not the PoC, but it was convenient in PoC to get run-time dimensions and enable/disable streams via the `RealSenseCamera` class. It also makes sense instead of going to the jvm-wrapper get it from the camera object instead, since it has those data members already, and just getting them would be as I said convenience.

cansik, 2019-03-15:
Well yes, I am not happy with the API at the moment and I will going to change it sooner or later again because I added a lot of new functionality in the jvm-wrapper, which is not included in the processing library yet.

cansik, 2020-01-07:
@smokhov If you are still interested in using the RealSense library from java, check out my latest wrapper project which ports the C# API to Java. Not all 2.29.0 features are supported yet, but most of the common ones are. There is also a test case on how to use it together with processing.

https://github.com/cansik/librealsense-java

closed 2020-02-05 by cansik