cansik/realsense-processing #12: How to detect face?

neciszhang, 2019-08-15:
I want to use it to detect people.But, i don't find the api to recognize it.Can you help me?

cansik, 2019-08-15:
Things, like detecting objects or tracking something by the SDK itself, is not supported atm. This is really just a very simple wrapper-library to get the streams from the camera.

I am working on implementing a library with more SDK functions, but this is still early alpha. Check out this issue for more information: https://github.com/cansik/realsense-processing/issues/10

And this is the link to the java wrapper with extended functionality:
https://github.com/cansik/librealsense/tree/master/wrappers/java

neciszhang, 2019-08-23:
Thanks you very much.

closed 2019-08-23 by neciszhang