cansik/realsense-processing #13: UnsatisfiedLinkError when running from source

mRamzii, 2019-11-23:
Hello,
First of all I would like to thank everyone that has contributed in this project, I am trying to run the source using the ColorizerTest example and IntelliJ IDE, when building, I get the following error :

`java.lang.UnsatisfiedLinkError: org.librealsense.Native.rs2CreateFrameQueue(I)J
	at processing.javafx.PSurfaceFX.lambda$0(PSurfaceFX.java:409)
	at java.lang.Thread.run(Thread.java:748)`

I believe this Is a problem linked to some native libraries not being compiled correctly ? I am trying to run this in Native Java not in Processing in order to implement more features and use it in a Java Project.

Thank you

cansik, 2020-01-07:
@mRamzii  Yes, you would have to rebuild it on windows yourself. But I now started a new wrapper project which contains the already prebuilt libraries for all platforms. It also supports way more of the official API and uses its structure. Check it out here:

https://github.com/cansik/librealsense-java

SamerDiab, 2020-01-29:
Hello @cansik , thank you for your work. What do you exactly mean by you have to rebuild it by yourself on windows? As i am getting this error, %1 is not a valid Win32 application when trying to load dll. I would like to use this project as i have a lot of work done based on in processing, and now i am trying to  prot the project to work outside of the processing ide.
Thanks.

cansik, 2020-01-29:
@SamerDiab "rebuilding" means building the java wrapper on your platform with the C API binding. Since I released the new version under [cansik/librealsense-java](https://github.com/cansik/librealsense-java) this step is not necessary anymore. The new library adds all the dependecies and methods.

To use processing outside of the Processing IDE, please head over to stackoverflow or the processing forum to get help (f.e. https://stackoverflow.com/questions/36765288/how-to-use-processing-3-on-intellij-idea).

SamerDiab, 2020-01-29:
Yes, and how exactly do you i have to build? Anything specific to the error mentioned? 
And are you planning to port the new library here?  Thank you.

cansik, 2020-01-30:
As I already wrote in my last comment:

> Since I released the new version under cansik/librealsense-java **this step is not necessary anymore**. The new library adds all the dependecies and methods.

The new library is already ported to Linux / Windows / OSX and can be downloaded in the [cansik/librealsense-java](https://github.com/cansik/librealsense-java) repo.

The code of Realsense-Processing is also updated to use the new library, but I did not release it as a prebuilt binary yet because there are still some API questions to not break the old API.

closed 2020-02-14 by cansik