cansik/realsense-processing #14: API change proposal for version 2.0

cansik, 2020-02-01:
I am currently trying to replace the old wrapper library with the new one, which should add the support for post processing and options. What you see here is just a **proposal** of how the API could look like. What I would like to get away from is the `true/false` enabling of streams in the constructor (very unintuitive). The only question for now is how to set options 🤔

```processing
RealSenseCamera camera = new RealSenseCamera(this);

void setup() {
  size(500, 500, FX2D);
  
  camera.enableRGB(640, 480);
  camera.enableDepth(640, 480, 30);
  camera.enableIR(640, 480, 30, IRStream.Second);
  
  // add processors
  camera.enableColorizer(ColorMap.BlackWhite);
  camera.enableAlign(Stream.Depth);
  
  camera.start();
}

void draw() {
  background(0);
  
  // how to set options?
  
  camera.readFrames();
  
  PImage rgb = camera.getRGBFrame();
  short[][] depth = camera.getDepthData();
  camera.getIRFrame(IRStream.Second);
}
```

cansik, 2020-02-01:
Well this API is now implemented. The question is, if it is necessary to build kind of a legacy API to support all the already written sketches? Or if it is fine to just release this new version (after a bit of testing) and hope that everyone updates their code.  🤷🏻‍♂️

oori, 2020-02-02:
No need for legacy API...  the difference is minor, and the examples clarify it. 
a. Bump the version tag up to v2.x.x
b. See the readme change I sent in PR

btw: you stopped at "_start reading the depth data by using_..." which **exactly** what I'm looking for now.    waiting for you next commit.   Thanks.

oori, 2020-02-02:
I've started testing the current master branch, so far works great!        I'll test it in the next day or two, and update if I bump into any issue.

oori, 2020-02-02:
Quick update,  all the examples work on D415 except these two fail with `RealSenseException`:

 1. `DisplayInfraredStream` -  doesn't have second infrared stream..  removing `IRStream.Second` fixes it. 
 2. `PointCloudViewer` - fails on `camera.start()`.    changing the `streamWidth/Height` to 640x480 resolves it.

Tested only on D415,    didn't test on other device yet.

cansik, 2020-02-03:
I was just testing it with a D430, but will give it a try later with a D415 (should be the same for the infrared streams). The only difference is, that the IR Stream on the D415 can also be read as RGB8. Maybe thats the problem.

The pointcloud sketch is just an example which shows how to extend the functionality with the java wrapper. I would like to add pointcloud creation to the API as well later.

oori, 2020-02-03:
the point cloud demo works,  it failed just because the resolution was wrong (set by D430 specs, but I used D415).     it works.

cansik, 2020-02-07:
@oori I tested all the examples today with a D415 and all of them worked (without adjusting something in the examples). Maybe there was a problem internally with your camera. Sometimes it starts to brick and do weird stuff, unplugging / replugging helps.

cansik, 2020-02-07:
I think we are good to go for a first version (the changes we discussed in #17 can be added later). I will release the new version this weekend.

cansik, 2020-02-14:
Released 2.1.0 to the contribution manager

closed 2020-02-14 by cansik

oori, 2020-02-15:
Thanks for your work.   great release!