cansik/realsense-processing #17: feat: setJsonConfiguration

oori, 2020-02-02:
**Ideal dev ux**:     Play and fine tune all the settings in realsense-viewer, and export the camera config to your processing app. 

**Inspiration**:  [here](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets)'s a good source for presets json.

oori, 2020-02-02:
What would be the correct way to use [setJsonConfiguration()](https://github.com/cansik/librealsense-java/blob/f9d0eba4172f92599712627eedb606706ed91104/src/main/java/org/intel/rs/device/AdvancedDevice.java#L56) in processing?         
I thought adding something like this in `RealSenseCamera`:
```
import org.intel.rs.device.AdvancedDevice;

public synchronized void setConfiguration(String jsonConfig) {
 AdvancedDevice advancedDevice = AdvancedDevice.fromDevice(device);
 advancedDevice.setAdvancedModeEnabled(true);
 // and catch if advanced mode isn't supported
 advancedDevice.setJsonConfiguration(jsonConfig);
}
```

should this be called on `start()`, where `device` is available?  or do you think one would require to re-`setJsonConfiguration()` during the app run, without `stop()` the camera.     For my use case, on `start()` makes sense,  but I can imagine funky stuff with manipulating this configuration on run.    (Not sure how much the intel can handle..)

([advanced mode doc](https://dev.intelrealsense.com/docs/intel-realsense-d400-advanced-mode))

cansik, 2020-02-05:
In my opinion I would add it before the `start()` method, because it changes the device.  The only problem is, that the device is maybe not set before starting the camera.

I think we have to look into how the device is added to the `RealSenseCamera`. Atm the device is handed over at start, but it is not directly possible to get all available devices.

oori, 2020-02-05:
The option is to overload `start(String jsonConfig)` isn't bad..  makes sense, but is limited (as explained [previous comment](https://github.com/cansik/realsense-processing/issues/17#issuecomment-581187446)).        What's your idea on revising the start sequence to support camera selection ?

cansik, 2020-02-05:
Overloading the start method with too many parameters is what I would like to avoid, because it leads to a tree of different configuration possibilities (with/without device, with/without json config, with/without something else).

We should add a method to list all devices and select one to hand over to the start method (as it is now). On the device itself it would then be possible to set the configuration. With this, we are still free to support a constructor, which is able to start the first am

```processing
RealSenseCamera cam = RealSenseCamera(this);

// get first device (normal / advanced)
Device device = cam.getDevice();
AdvancedDevice advancedDevice = cam.getAdvancedDevice();

// get all devices
Device[] devices = cam.getDevices();

// do someting (like setting config)
advancedDevice.setJsonConfiguration(jsonConfig);

// start camera with default getDevice()
cam.start();

// start camera with device
cam.start(advancedDevice);

// start camera by using serial number (to always get the same camera)
cam.start(String serialNumber);
```

cansik, 2020-02-05:
The problem with getting the device is the following:

Getting a device from the deviceList requires to call `rs2_create_device()`. That means, that the device is created and not available anymore for another source. To read information of a device, we have to kind of "lock" it. In case the user won't call `close()` on it, we will "loose" the device until the applications runs.

oori, 2020-02-06:
So, the issue is "only" with listing the devices `getDevices()`?        I mean, the above could be implemented, as is, with the current mechanism (first device found is selected).

cansik, 2020-02-07:
Yes, I could implement those things as a constructor.

cansik, 2020-02-11:
Ok, it seems that I have implemented something wrong: The json config can not be applied before running the device, it has to be applied after that. I have fixed it in the last commit https://github.com/cansik/realsense-processing/commit/f11dc301b7ab87f7827375be4346ebb5a9f23cda

oori, 2020-02-11:
Thanks!   I will update my code later this week with these updates.   will update here.

cansik, 2020-02-14:
This is fixed and released to the contribution manager today.

closed 2020-02-14 by cansik