cansik/realsense-processing #22: Getting "A library used by this sketch is not installed properly" error

stevenjohnson83, 2020-02-14:
"A library relies on native code that's not available.
Or only works properly when the sketch is run as a 32-bit application.
UnsatisfiedLinkError: Can't load library: /Users/this user/Documents/Processing/libraries/RealSenseProcessing/examples/CameraAvailable/code/native/osx-x64/librealsense2.dylib"

Its on a fresh instal on osx 10.15.3 
Processing 3.5.4
Tried the library from the contribution manager and manually placing the in the library path. Both threw these errors when trying to run the examples

cansik, 2020-02-14:
Could you please further explain what you mean by:
> manually placing the in the library path

Either you download the library through the contribution manager or you add the jar directly to the sketch. Doing both does not really make sense.

stevenjohnson83, 2020-02-18:
Feel free to close this, re downloaded the library through the contribution manager and works fine now. 
But to explain. 
"Either you download the library through the contribution manager"
Which i did, but was getting the above error.
So i downloaded the the latest build, replaced the contribution library with the build and examples, tried dropping the jar files (confirmed they were in the code subfolder) but was still getting the error. 

Seems resolved now.

closed 2020-02-18 by stevenjohnson83