cansik/realsense-processing #24: Colorizer Option

hartonoda, 2020-02-17:
Hello, first of all thank you so much for this library, 
Im trying to disabled the Histogram Equalization Enabled through the json configuration but it gave me error, then I tried to check if I can edit the source of the library but I couldnt find the list of colorizer property that I can access. Would you know by any chance?
thanks in advance!

cansik, 2020-02-17:
The json config is just for the device settings, not for the post processing pipeline. As mentioned [here](https://github.com/IntelRealSense/librealsense/issues/1674#issuecomment-387361411) it is possible to set the colorizer histogram. At the moment I did not expose the colorizer processor in the library, but I just added a getter for you in the [latest](https://github.com/cansik/realsense-processing/commit/0988baef3bc534391268bde2899983e1ccd1b00b) commit. The options are stored in the `Option` enum, so changing the exposure setting should work like this:

```java
camera.getColorizer().setOption(Option.HistogramEqualizationEnabled, 0f);
```

hartonoda, 2020-02-17:
Thank you so much Florian! do you maybe have list the colorizer option parameter?
also do you know how can I set min range and max range?
thanks again!

cansik, 2020-02-18:
@hartonoda I would suggest to look into the RealSense library documentation, maybe there is somewhere a list. Otherwise you could also just loop over each option that is supported by the colorizer processing block:

```java
import org.intel.rs.option.CameraOption;
import org.intel.rs.option.SensorOptions;

camera.enableColorizer();

// list all possible options
for(CameraOption option : camera.getColorizer().getBlock().getOptions()) {
    System.out.println(option.getKey() + ": " + option.getDescription());
}
```

Returns following list:

```
VisualPreset: Preset depth colorization
FramesQueueSize: Max number of frames you can hold at a given time. Increasing this number will reduce frame drops but increase latency, and vice versa
ColorScheme: Color map
HistogramEqualizationEnabled: Perform histogram equalization
MinDistance: Min range in meters
MaxDistance: Max range in meters
StreamFilter: Stream type
StreamFormatFilter: Stream format
StreamIndexFilter: Stream index
```

closed 2020-09-21 by cansik