cansik/realsense-processing #25: Is it possible to use more than one camera simultaneously?

BellLabsEAT, 2020-03-04:
I was wondering if this feature has been updated?

cansik, 2020-03-04:
Yes, should be possible now. The problem is that it's not as easy as using an **index** to select the  desired RealSense device. Because of that I did not document it publicly. But I can show how to use it through an example.

cansik, 2020-03-04:
Check out the [Multi Color Stream example](https://github.com/cansik/realsense-processing/blob/master/examples/MultiCameraColorStream/MultiCameraColorStream.pde) I've just uploaded. It uses the serial number to start the desired camera.

cansik, 2020-03-06:
@BellLabsEAT I have added two examples, one just shows how to [start multiple cams](https://github.com/cansik/realsense-processing/blob/master/examples/MultiCameraColorStream/MultiCameraColorStream.pde) with their serial numbers, the other shows how to use the [devices querying](https://github.com/cansik/realsense-processing/blob/master/examples/AdvancedDeviceHandling/AdvancedDeviceHandling.pde).

closed 2020-03-06 by cansik