cansik/realsense-processing #26: Support for T265?

fig8studio, 2020-03-29:
Would it be possible to make this available for tracking with Intel T265?

cansik, 2020-03-29:
Yes, but for that we would first have to adapt the base java library to support pose frames. Would be great if you could add it and create a pull request.

cansik, 2021-01-19:
Since I do not have a camera, it would make sense that you try out the following code and give me feedback if it works (should show the x rotation in the center of the screen): [PoseStreamTest.java](https://github.com/cansik/realsense-processing/blob/master/src/test/java/ch/bildspur/realsense/test/PoseStreamTest.java)

TheBricktop, 2021-09-11:
I can download and try it on my machine.

TheBricktop, 2021-09-18:
Got an error when ported Your code to processing 4b1

[image]

cansik, 2021-09-19:
Ok thanks for reporting...I guess I need a camera to test it out further. It's hard to develop for hardware that you can not test on, sorry.

TheBricktop, 2021-09-20:
If id set up a pc rigged with t265 and allowed You to controll it remotely using anydesk would that help?

cansik, 2021-09-20:
If you have a spare one, I could give it a try. But I can't guarantee success :)

TheBricktop, 2021-09-21:
I can set it up today and send You computer adress

cansik, 2021-09-21:
Yes, if you look at the [method implementation](https://github.com/cansik/realsense-processing/blob/b3ee9970ca94b374d192aff30d96bfd9c1ca7445/src/main/java/ch/bildspur/realsense/RealSenseCamera.java#L161-L164), I am using the same configuration.

```java
public void enablePoseStream() {
    // this does not work like the video streams (only uses type and format)
    poseStream.init(Stream.Pose, Format.SixDOF);
    config.enableStream(Stream.Pose, Format.SixDOF);
}
```

TheBricktop, 2021-09-21:
Camera is found allright, it only throws error when camera.start() is called, wonder what is causing it.