cansik/realsense-processing #27: data quota error (LFS bandwidth)

jeremydouglass, 2020-06-08:
This repo is tracked by archive-for-processing. Every few weeks it attempts to fetch any recent changes from upstream. However, recently this has been failing. I inspected the logs, and this is what I found:

> Error downloading object: native/linux-64/libnative.so (3e03155): Smudge error: Error downloading native/linux-64/libnative.so (3e03155d3f3b152191f92eb9fac2af62976f73a63a4cefe716a103eea702171f): batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.

I don't know what kind of bandwidth you need for LFS, but I'd be happy to contribute a small [gift card](https://github.myshopify.com/products/gift-card) -- and if the archive fork has any significant roll in consuming your repo bandwidth then we can fix that.

cansik, 2020-06-08:
Wow, thanks for pointing that out. I was not aware that they lock the access also by bandwidth (the space quota already overshot its limits).

[image]

Currently I do not have any LFS tracked file in the head commit because I switched to a maven based solution for distributing these large files. I will have a look how I can clean the repository with BFG Repo-Cleaner and remove the large files in the history. If it is possible, I'll upload them to the release's for people who like to dig into history.

Even though I appreciate this gesture very much, it is not necessary to support me with a gift card. ✨Thank you!

jeremydouglass, 2020-06-08:
> clean the repository with BFG Repo-Cleaner and remove the large files in the history. If it is possible, I'll upload them to the release's for people who like to dig into history.

This sounds ideal, thank you so much.

cansik, 2020-06-13:
It's super annoying because I do not have access to the large files anymore (because of the bandwidth). So fixing it is quite a painful task. Github sent me all the names of the repos that have LFS enabled (and their sizes) and told me to delete the files on my side (with bfg cleaner).

I wrote a [rm-lfs.sh](https://gist.github.com/cansik/ad9bdc68d9513016e963f71f722d8a50) script to do this task for all the repos, but currently I am not able to push the changes because of the bandwidth limitation of LFS 🤷🏻‍♂️

Maybe I have to wait till the start of a new month to use my own quota...but this is really ridiculously done by Github.

cansik, 2020-06-14:
Last commit ca68b6177064207c99e313e3f0af42747345da8b removed all LFS files and altered the git history. It should be possible now to just checkout the repository, without downloading any git LFS files or limiting quotas.

Would be great if you could test it on your side too.

jeremydouglass, 2020-06-14:
Thanks, @cansik -- when I tried to update the existing fork, I got the same error. However, once I dropped the fork and re-forked based on your rewritten repo, everything was clean!

closed 2020-06-14 by jeremydouglass