cansik/realsense-processing #28: Does this work on linux-armhf or linux-arm64?

DieProgrammIDE, 2020-06-11:
Does this work on linux-armhf or linux-arm64?
I would like to use it on a Raspberry Pi 4

cansik, 2020-06-11:
At the moment I would say no, because there are only libraries for x86 and x64 packed. We would have to add a new platform build to the [preset library](https://github.com/bytedeco/javacpp-presets/blob/master/librealsense2/cppbuild.sh#L38).

Or build it realsense2 and it's java binding directly on the raspberry.

DieProgrammIDE, 2020-06-20:
?

cansik, 2020-06-23:
I can not give you instructions or support regarding this, because I do not have the time at the moment. Maybe someone else has more time to help you. I will flag it as help wanted.

cansik, 2020-09-22:
Maybe ARM support is coming soon: https://github.com/bytedeco/javacpp-presets/issues/945

cansik, 2021-01-19:
Linux-armhf and linux-arm64 are now supported. Release should follow today.

closed 2021-01-19 by cansik