cansik/realsense-processing #29: Example Sketches

flowv, 2020-08-04:
Hi Forian,

I just downlaoded the latest release of the deepvision library. I all exoted to test it out, but i can't get the examples to work. Specifically, it't the "FaceAndEmotionWebCam", line 43 where processing tells me that the "Width(0) and Height(0) cannot be <=0". Could you kindly give me a hint how to fix it? 
Thanks you so much!

cansik, 2020-08-04:
1. Could you give me a bit more information, at least library version & operating system please.
2. Usually this exception means that the camera did not send a frame yet. I recommend to check the frame first, before passing it into the deep-vision framework (have a look at the example below).
3. Next time, please post issues related to the deep-vision framework to its repository.

```pde
if (cam.available()) {
  cam.read();
}

// this guard leaves the draw method
// until a frame has arrived
if (cam.width == 0) {
  return;
}

image(cam, 0, 0);
```

flowv, 2020-08-04:
Oh, that's it! It works now! Thank you so much!
Nice framework, it put me much younger!

closed 2020-08-04 by flowv