cansik/realsense-processing #30: Disable Auto-Exposure

Sensebellum, 2020-09-21:
Great work on this library!

Curious if it is possible to turn off the auto-exposure using this library?

How would that be coded? I see the filters and those are helpful. But I did not see the option to do anything to auto exposure.

Thank you!

cansik, 2020-09-21:
Hi, well this is still an open issue #18 but I have already implemented it in the JavaCV RealSense2 FrameGrabber:
[bytedeco/javacv/RealSense2FrameGrabber.java#L365](https://github.com/bytedeco/javacv/blob/master/src/main/java/org/bytedeco/javacv/RealSense2FrameGrabber.java#L365)

With this library I had problems, because some of the settings you can't change after starting the pipeline.

What I currently suggest, is just using the json configuration, have a look here:
https://discourse.processing.org/t/intel-realsense-library-for-processing/8143/25

cansik, 2020-09-21:
@Sensebellum I have implemented sensor options now, please head over to the release section and download the [pre-release 2.2.0](https://github.com/cansik/realsense-processing/releases/tag/2.2.0). Would be great if you could quickly try it out, maybe with [this example](https://github.com/cansik/realsense-processing/blob/master/examples/SetSensorOptions/SetSensorOptions.pde) or here the [readme section](https://github.com/cansik/realsense-processing#sensor-options).

Sensebellum, 2020-09-21:
Awesome! I will give that a try later. 

Looks like there is a slightly different way of loading the JSON file as someone used this way of doing it with the "param":

`String param = "\"controls-laserstate\": \"off\"";
String jsonConfig = "{" + param + "}";`

I also look forward to trying the new version with sensor options.

I see this as a way to disable auto-exposure.

`camera.getRGBSensor().setOption(Option.EnableAutoExposure, 0.0f);`

Could one apply that to the depth like so?

'camera.getIRSensor().setOption(Option.EnableAutoExposure, 0.0f);'

Thank you!

cansik, 2020-09-21:
Yes, `getDepthSensor()` is what you are looking for.

Sensebellum, 2020-09-22:
"The function getDepthSensor() does not exist."

Sensebellum, 2020-09-22:
"The function getRGBSensor() does not exist."

Sensebellum, 2020-09-22:
Ahhh yes of course I need to download the new library. Works now! Nice work!!!

Sensebellum, 2020-09-22:
This feature was a game changer thank you so much!!!

cansik, 2020-09-22:
I have released 2.2.0 for everyone, so I will close this issue.

closed 2020-09-22 by cansik