cansik/realsense-processing #31: Running out of memory?

Sensebellum, 2020-09-21:
Just getting to know this project long term and I found it crashes between 5-20 minutes in. 

Basically just trying to render the Depth Stream and do some coloring over it before sending it to MadMapper.

Some of the errors reference the intel.something something and I am curious if there is a way to resolve this or do something else to fix it?

Thanks!

ERROR CODES

> java.lang.OutOfMemoryError: Cannot allocate new IntPointer(1): totalBytes = 8, physicalBytes = 7249M

> Spout 64bit v2.0.6.0 - Java 1.8.0_202
Created sender 'Spout Processing' (848x480)
Spout initialized texture sharing
java.lang.RuntimeException: java.lang.OutOfMemoryError: Cannot allocate new IntPointer(1): totalBytes = 8, physicalBytes = 7249M
	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:412)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.OutOfMemoryError: Cannot allocate new IntPointer(1): totalBytes = 8, physicalBytes = 7249M
	at org.bytedeco.javacpp.IntPointer.<init>(IntPointer.java:86)
	at org.intel.rs.stream.StreamProfile.<init>(StreamProfile.java:16)
	at org.intel.rs.frame.Frame.getProfile(Frame.java:101)
	at org.intel.rs.frame.FrameList.getFirstOrDefault(FrameList.java:77)
	at org.intel.rs.frame.FrameList.getFirstOrDefault(FrameList.java:72)
	at org.intel.rs.frame.FrameList.getDepthFrame(FrameList.java:87)
	at ch.bildspur.realsense.RealSenseCamera.readFrames(RealSenseCamera.java:292)
	at IntelDepth3.draw(IntelDepth3.java:69)
	at processing.core.PApplet.handleDraw(PApplet.java:2475)
	at processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:866)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
	at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
RuntimeException: java.lang.OutOfMemoryError: Cannot allocate new IntPointer(1): totalBytes = 8, physicalBytes = 7249M
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:759)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
Caused by: java.lang.OutOfMemoryError: Physical memory usage is too high: physicalBytes (7249M) > maxPhysicalBytes (7247M)
	at org.bytedeco.javacpp.Pointer.deallocator(Pointer.java:659)
	at org.bytedeco.javacpp.Pointer.init(Pointer.java:127)
	at org.bytedeco.javacpp.IntPointer.allocateArray(Native Method)
	at org.bytedeco.javacpp.IntPointer.<init>(IntPointer.java:78)
	... 20 more

SKETCH

`import ch.bildspur.realsense.*;
import ch.bildspur.realsense.type.*;
import ch.bildspur.realsense.*;
import ch.bildspur.realsense.type.ColorScheme;
import org.intel.rs.device.AdvancedDevice;

import spout.*;
Spout spout;

int b = 0;
int adjust = 1;

RealSenseCamera camera = new RealSenseCamera(this);

////////////////////////////////////////////////////////////////////////////////////

void setup()
{
  size(848, 480, P3D);
  textureMode(NORMAL);
  smooth();

  spout = new Spout(this);
  spout.createSender("Spout Processing");

  // enable depth stream
  camera.enableDepthStream(848, 480);

  // enable colorizer to display depth
  //camera.enableColorizer(ColorScheme.Classic);
  camera.enableColorizer(ColorScheme.WhiteToBlack);
  //camera.enableColorizer(ColorScheme.Quantized);
  camera.addThresholdFilter(1.1, 1.44);
  camera.addTemporalFilter(0.3f, 19, PersistencyIndex.ValidIn2_Last4);

  camera.start();
}

////////////////////////////////////////////////////////////////////////////////////

void draw()
{
  background(0);

  // read frames
  camera.readFrames();

translate(0, -15, 190);
  // show color image
  image(camera.getDepthImage(), 0, 0);
  
    if (mousePressed == true) {
    println(brightness(get(mouseX, mouseY)));
  }
  
  // Do Something cool...
  
  filter(BLUR, 1);
  
//Colorrr();

  
// Send out via Spout
  spout.sendTexture();
}`

cansik, 2020-09-22:
Next time, please tell me at least on which operating system your are running it :) Of course, spout is a windows technology, but it would still be helpful.

Regarding the "memory leak", I have run your test on MacOS without any problems. Maybe it's because of spout?

[image]

Sensebellum, 2020-09-22:
@cansik Of course! 

I am running Windows 10 here. Intel NUC Hades Canyon as the build.
Processing 64-bit.

I think I found what the problem was and this is the BLUR filter. I turned that off and it all runs super smooth. Boots great. No crashes. Maybe there is a problem with BLUR in the Win 10 distro of Processing?

Not sure if that is to be expected.

Thanks for digging into this though you rock!

closed 2020-09-22 by cansik