cansik/realsense-processing #32: Intel L515 support

oori, 2020-09-23:
`RealSense: No device found!` with the new L515 camera.
Camera does work with `realsense-viewer` 2.38.1,    and this repo uses 2.29.   could that be the problem?

BTW, on same setup: 
D415 works great.
SR305 also works fine (_although fails on pointcloud demo_)

cansik, 2020-09-23:
Yes, I am currently working on that, but first we have to update the native base libraries: https://github.com/bytedeco/javacpp-presets/pull/946

If you are working on MacOS, you can replace the jars with the following and it should run with the Intel L515: [librealsense2-java-2.38.1.zip](https://github.com/cansik/realsense-processing/files/5268789/librealsense2-java-2.38.1.zip)

oori, 2020-09-23:
Thanks a lot.
I do development on MacOS, but eventually (in 3-4 weeks) it needs to runs on Windows 10.  I'll track your repo for updates, and will be happy to assist with testing.

oori, 2020-09-30:
Tried it on Mac now, I get:
```
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x0000000135ad5cd2, pid=41952, tid=0x000000000000f053
#
# JRE version: Java(TM) SE Runtime Environment (8.0_202-b08) (build 1.8.0_202-b08)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.202-b08 mixed mode bsd-amd64 compressed oops)
# Problematic frame:
# C  [librealsense2.2.38.dylib+0x383cd2]  _ZN12librealsense10uvc_sensor4openERKNSt3__16vectorINS1_10shared_ptrINS_24stream_profile_interfaceEEENS1_9allocatorIS5_EEEE+0x334
```

Full log here: [hs_err_pid41952.log](https://github.com/cansik/realsense-processing/files/5307719/hs_err_pid41952.log)

oori, 2020-09-30:
Do note, that on `realsense-viewer` 2.38.1 this L515 crashes the app occasionally.  as if bus is overloaded. most commonly, if you try to start the `Motion Module` (even by itself, without the other modules).    So this `SIGSEGV` reminded me of `segmentation fault`.

```
 30/09 22:11:24,928 WARNING [0x7000098c0000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Operation timed out, number: 60
 30/09 22:11:25,073 WARNING [0x7000098c0000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Operation timed out, number: 60
[1]    43369 segmentation fault  realsense-viewer
```

oori, 2020-09-30:
Maybe related to this? https://github.com/IntelRealSense/librealsense/issues/6925

Similar here:
https://github.com/cansik/realsense-processing/blob/153f2bf3f7818826bc6c4e7a4278a4d98c5de569/src/main/java/ch/bildspur/realsense/RealSenseCamera.java#L450

oori, 2020-10-11:
> Yes, I am currently working on that, but first we have to update the native base libraries: [bytedeco/javacpp-presets#946](https://github.com/bytedeco/javacpp-presets/pull/946)

@cansik   I see it was merged a couple of days ago.   any news about getting this into this library?  Thanks.

cansik, 2020-10-11:
Thank you, I have already seen it.

1.5.5 is currently snapshot, and because gradle is not great at handling maven snapshot releases, I will wait until javacpp-presets 1.5.5 are officially released. Then it should be a nobrainer, just editing the gradle files of the java warpper and re-building the realsense-processing library.

You could of course do that already now... :)

oori, 2020-10-11:
Thanks.
I rebuilt `librealsense-java` with `1.5.5-SNAPSHOT` / `2.38.1-1.5.5-SNAPSHOT`, but two tests fail.
```
> Task :test FAILED

org.intel.rs.test.DeviceTest > deviceCountTest FAILED
    org.intel.rs.util.RealSenseException at DeviceTest.java:20
    java.lang.NullPointerException at DeviceTest.java:25

org.intel.rs.test.DeviceTest > setJsonConfigTest FAILED
    org.intel.rs.util.RealSenseException at DeviceTest.java:20
    java.lang.NullPointerException at DeviceTest.java:25

14 tests completed, 2 failed
```

```
org.intel.rs.util.RealSenseException: null pointer passed for argument "callback"
	at org.intel.rs.util.RealSenseUtil.checkError(RealSenseUtil.java:13)
	at org.intel.rs.Context.<init>(Context.java:40)
	at org.intel.rs.test.DeviceTest.createContext(DeviceTest.java:20)
```

I will look into it again later. 

@cansik,  Do you have an idea/direction regarding the macOS crash with `SIGSEGV`?  see comments above. Thanks again!

oori, 2020-10-18:
friendly bump: 2.39.0 is already out for some time

cansik, 2020-10-19:
Thanks! If you would like to contribute, you could already bump the javacpp-presets to 2.39.0. I am still waiting for the release of 1.5.5

cansik, 2021-01-19:
Intel RealSense 515 is now supported. A prebuilt release should follow today.

closed 2021-01-19 by cansik