cansik/realsense-processing #33: getDistance() for Multiple Camera Feeds

danielleamya, 2021-01-18:
Hi,

I have been working with two Realsense cameras to use the depth streams for blob tracking. The depth data is a bit noisy at times so I am looking to use the getDistance() function to be able to cross-confirm what should and should not be considered a blob. For some reason I am having some issues calling getDistance() with the second camera (the sketch constantly crashes). Is there any way to get this to work?

Thanks,
Danielle

cansik, 2021-01-18:
Multicam support is still a bit buggy....on which system are you running it (Mac / Windows / Linux) and is a jvm-crash logfile written (something like `hs_err_pid2075.log`)?

danielleamya, 2021-01-18:
Hi, I am using a Windows machine; a jvm-crash logfile is written (attached here [hs_err_pid6900.log](https://github.com/cansik/realsense-processing/files/5832069/hs_err_pid6900.log)).

Here is a minimal example for your testing. 
`import ch.bildspur.realsense.*;
import ch.bildspur.realsense.type.*;
import ch.bildspur.realsense.*;
import ch.bildspur.realsense.type.*;
import oscP5.*;
import netP5.*;
import websockets.*;

RealSenseCamera cam1 = new RealSenseCamera(this);
RealSenseCamera cam2 = new RealSenseCamera(this);

void setup() {
  size(640, 960);
  // enable depth stream
  cam1.enableDepthStream(640, 480);
  cam2.enableDepthStream(640, 480);

  // enable colorizer to display depth
  cam1.enableColorizer();
  cam2.enableColorizer();

  cam1.start("027422070571");  // Insert serial number of camera here!   944622072933 944622073267
  cam2.start("834412070190");
}

void draw() {
  background(0);

  cam1.readFrames();
  cam2.readFrames();
  image(cam1.getDepthImage(), 0, 0);
  image(cam2.getDepthImage(), 0, 480);
}

void mousePressed() {
  float distance;
  if(mouseY < height/2){
    distance = cam1.getDistance(mouseX, mouseY);
  } else {
    distance = cam2.getDistance(mouseX, mouseY);
  }
  println(distance);
}`

cansik, 2021-01-18:
Thanks, I will investigate on it tomorrow! So just for clarification: just `cam2.getDistance(mouseX, mouseY)` crashes?

danielleamya, 2021-01-18:
Thanks so much!
Yes, that line `cam2.getDistance(mouseX, mouseY)` causes the sketch to crash. Without that line, everything proceeds just fine.

cansik, 2021-01-19:
So I could test it and it seems it's a bug in your code which is easy to fix: In `mousePressed` you check on which camera image you currently are and than use `getDistance` from the underlaying camera. This is fine, but the `mouseY` coordinates will be between `480 to 920`  for the second camera. To avoid reading the distance of a pixel that is not available, you can just subtract `480` from the `mouseY` value and get the real image range.

```processing
void mousePressed() {
  float distance;
  if(mouseY < height/2){
    distance = cam1.getDistance(mouseX, mouseY);
  } else {
    distance = cam2.getDistance(mouseX, mouseY - 480);
  }
  println(distance);
}
```

I did the same thing in my test sketch (but horizontally). Hope this works for you.
But of course, this could be checked by the realsense library beforehand and I have already implemented a check if the coordinates are corresponding to the image size or not. Will be available in the next release.

cansik, 2021-01-24:
@danielleamya Would be great if you could give me a feedback if that solves the problem for you?

BellLabsEAT, 2021-01-24:
Oh sorry, I thought I responded to this already, my apologies. This has been working perfectly for me, thanks so much!

closed 2021-01-24 by cansik