cansik/realsense-processing #37: MacOS Monterey update "failed to set power state"

owenmcateer, 2022-02-04:
Hello,
Since updating my Mac to OSX Monterey I can no longer get the RealSense camera to connect.
Running my own scrips or the example I get the same error message:  
`RealSenseException: failed to set power state`

I have the D415 camera. It can be found as a USB device in system info:
[image]

Any ideas or tips on what I should try to get it working?
Thanks

cansik, 2022-02-04:
There are still a lot of problems with Monterey. I recommend to check out this issue thread for more infos and specifically this post because it has to do with the [realsense-processing](https://github.com/IntelRealSense/librealsense/issues/9916#issuecomment-1026803051) library.

cansik, 2022-09-21:
I am currently playing around with arm64 builds of the realsense library. If you would like to try out my pre-built version, download it here: [apple-silicon/RealSenseProcessing.zip](https://github.com/cansik/realsense-processing/releases/download/apple-silicon/RealSenseProcessing.zip)

CaseyJScalf, 2022-11-10:
Dusting off my camera from last year and I am surprised to learn it no longer works. I was just about to make an issue report and found this. Has there been any success? Seriously though these updates... smh

I am on a 2015 MBP running OS Monterey 12.6

How do we fix this?

Thanks for any help!

CaseyJScalf, 2022-11-10:
Just an update here but I tried updating the library to the link above and the same issue is present "Failed to set power state".

I also tried the latest 2.4.3 and same issue.

I did this for both Processing 3.5.4 and Processing 4.01.

Any other ideas on how to get this back working?
[image]

cansik, 2022-11-11:
Hi @CaseyJScalf , this is still an open issue. As already mentioned, Intel does not work on the MacOS support anymore. So there will not be any updates regarding that issue, until they decide differently. It is possible to use the camera with `sudo` under OSX, but this is not great and often fails. It is a real pity.

Windows and Linux are still supported and work out of the box.

CaseyJScalf, 2022-11-11:
Hell and thank you for the update. This is a true shame for these cameras are the best and to not have them supported in the Mac OS world is actually kind of wild - or is this issue only for the Processing side of things?

I luckily had a windows machine and was able to get it running on there no problem.