cansik/realsense-processing #39: Can't access IR image w/D415 camera

cacheflowe, 2023-03-13:
Thank you for this library - I've used it on many projects in recent years 🙏 

I've realize that I've never been able to use the IR image, but I would like to :)

Any time I try, I get this error, and the program crashes:

```
RealSenseException: Couldn't resolve requests
```

Here's the full example, using the latest Realsense Processing library (2.4.3), and the latest Processing (4.2) on Windows 11. There's no difference if I don't specify which IR stream it should use. This is basically the demo in the README:

```
import ch.bildspur.realsense.*;
import ch.bildspur.realsense.type.*;

RealSenseCamera camera = new RealSenseCamera(this);

void setup()
{
    size(1280, 480);
    
    camera.enableColorStream(640, 480, 30);
    camera.enableIRStream(640, 480, 30, IRStream.Second);
    camera.start();
}

void draw()
{
    background(0);
    
    // read frames
    camera.readFrames();
    
    // show images
    image(camera.getColorImage(), 0, 0);
    image(camera.getIRImage(IRStream.Second), 640, 0);
}
```

Let me know if I can help test anything, and thank you!

cansik, 2023-03-13:
Hi, I do not own a D415 so it's not easy to test. I already worked with them so it should display the IR image, but maybe due to framework changes something broke. Maybe I get access to one on Wednesday so I can further investigate.

cansik, 2023-03-13:
And btw. really nice installations ✨I like the patterns of Melter a lot.

cacheflowe, 2023-03-15:
Thank you! 🙏  Your tools enable my work, and I'm always impressed with what you create. 

I would mail you one of my cameras from the USA if you'd mail it back!

cansik, 2023-03-23:
@cacheflowe Hi, it took me longer than expected to get a D415, but I have tested it now on a Windows 11 with the most recent version. And it seems to work without any issues, example sketches as well as the one you provided. Have you checked if you are using the correct USB cable and have you already changed the USB port?

cacheflowe, 2023-03-26:
Well now I feel bad that you made this effort. I installed everything on a different machine and it works... I have no idea what the difference is. All of the cameras & cables work on one machine, but not the other, and it's *only* the IR image that doesn't show up. Everything else looks normal. Maybe you should close the issue, and I'll add to this thread if I figure out the problem with my environment. Thank you for taking the time to look at this 🙏

cansik, 2023-03-27:
No problem, RealSense cameras can be very stubborn. If you find out anything else, it would be great if you could describe it here. Thanks!

closed 2023-10-25 by cacheflowe